FreeRTOS/coreMQTT #105: Log functions cause compiler warnings

boraozgen, 2020-11-05:
Hi everybody,

First of all very good job with the new version, the quality seems to be improved massively. I hope this will be the last major version I will be porting :)

I am porting the SDK to Mbed OS, and realized that the log functions cause compiler warnings. My GCC ARM version is `9 2020-q2-update`.

There might be some I missed, but here is the what I caught:

```
[Warning] core_mqtt.c@608,25: format '%d' expects argument of type 'int', but argument 4 has type 'int32_t' {aka 'long int'} [-Wformat=]
[Warning] core_mqtt.c@623,25: format '%d' expects argument of type 'int', but argument 4 has type 'int32_t' {aka 'long int'} [-Wformat=]
[Warning] core_mqtt.c@623,25: format '%d' expects argument of type 'int', but argument 6 has type 'int32_t' {aka 'long int'} [-Wformat=]
[Warning] core_mqtt.c@635,21: format '%u' expects argument of type 'unsigned int', but argument 4 has type 'uint32_t' {aka 'long unsigned int'} [-Wformat=]
[Warning] core_mqtt.c@717,25: format '%d' expects argument of type 'int', but argument 4 has type 'int32_t' {aka 'long int'} [-Wformat=]
[Warning] core_mqtt.c@734,25: format '%d' expects argument of type 'int', but argument 4 has type 'int32_t' {aka 'long int'} [-Wformat=]
[Warning] core_mqtt.c@734,25: format '%d' expects argument of type 'int', but argument 6 has type 'int32_t' {aka 'long int'} [-Wformat=]
[Warning] core_mqtt.c@786,25: format '%d' expects argument of type 'int', but argument 4 has type 'int32_t' {aka 'long int'} [-Wformat=]
[Warning] core_mqtt.c@813,21: format '%d' expects argument of type 'int', but argument 4 has type 'uint32_t' {aka 'long unsigned int'} [-Wformat=]
[Warning] core_mqtt.c@854,24: format '%d' expects argument of type 'int', but argument 4 has type 'int32_t' {aka 'long int'} [-Wformat=]
[Warning] core_mqtt.c@859,25: format '%d' expects argument of type 'int', but argument 4 has type 'int32_t' {aka 'long int'} [-Wformat=]
[Warning] core_mqtt.c@949,25: format '%d' expects argument of type 'int', but argument 5 has type 'int32_t' {aka 'long int'} [-Wformat=]
[Warning] core_mqtt.c@1388,21: format '%d' expects argument of type 'int', but argument 4 has type 'int32_t' {aka 'long int'} [-Wformat=]
[Warning] core_mqtt.c@1406,29: format '%d' expects argument of type 'int', but argument 4 has type 'int32_t' {aka 'long int'} [-Wformat=]
[Warning] core_mqtt.c@1775,25: format '%d' expects argument of type 'int', but argument 4 has type 'int32_t' {aka 'long int'} [-Wformat=]
[Warning] core_mqtt.c@1866,25: format '%d' expects argument of type 'int', but argument 4 has type 'int32_t' {aka 'long int'} [-Wformat=]
[Warning] core_mqtt.c@2001,25: format '%d' expects argument of type 'int', but argument 4 has type 'int32_t' {aka 'long int'} [-Wformat=]
[Warning] core_mqtt.c@2061,25: format '%d' expects argument of type 'int', but argument 4 has type 'int32_t' {aka 'long int'} [-Wformat=]
[Warning] core_mqtt.c@2111,25: format '%d' expects argument of type 'int', but argument 4 has type 'int32_t' {aka 'long int'} [-Wformat=]
```

dan4thewin, 2020-11-19:
Hi, @boraozgen .  Thanks for reporting this.  I'm not getting these warnings, even using the same version of gcc for ARM, but I think we might solve it using `%ld` and a cast to long, like so:

    LogDebug( ( "Sent %ld bytes of PUBLISH header.", ( long ) bytesSent ) );

Would you like to try that and see if it works for you?  If so, we can make a PR to change all the formats.

boraozgen, 2020-11-19:
Yes, `%ld` and `%lu` fixes them. It is interesting though that the warnings do not show up. I use the compiler flags `-Wall` and `-Wextra`, which should imply `-Wformat`, which should turn on these checks.

dan4thewin, 2020-11-20:
Can you give me the full command line you use?  In the meantime, we'll start on a PR.

boraozgen, 2020-11-23:
It is hard to tell the exact command since I am using the build tooling of Mbed.

dan4thewin, 2020-11-24:
The PR with the changes has been merged: https://github.com/FreeRTOS/coreMQTT/pull/113

Please let us know if we can help with any further issues.

closed 2020-11-24 by dan4thewin