FreeRTOS/coreMQTT #111: Publish not checking for entire payload to have been sent

snap40ben, 2020-11-19:
https://github.com/FreeRTOS/coreMQTT/blob/4b8674a80627e8a210a38e6fa2f5fc2731ede0e5/source/core_mqtt.c#L632

I think this should be comparing with `bytesToSend` instead of `0`.

On a side note `bytesRemaining` is assigned twice at the beginning of the function.

dan4thewin, 2020-11-24:
Hi, @snap40ben .  It appears to me that if `totalBytesSent` is greater than zero, then it must be equal to `bytesToSend`.  Otherwise, the prior while loop would continue.  Would you like to submit a PR to fix the extra assignment?  We welcome your participation!

snap40ben, 2020-11-25:
> Hi, @snap40ben . It appears to me that if `totalBytesSent` is greater than zero, then it must be equal to `bytesToSend`. Otherwise, the prior while loop would continue. Would you like to submit a PR to fix the extra assignment? We welcome your participation!

Hi @dan4thewin, if `pContext->transportInterface.send` fails (say wifi connection goes down) in the middle of a transmission requiring several calls to `pContext->transportInterface.send`, then `totalBytesSent` could be >0 but different from `bytesToSend`.

Am I understanding it right?

dan4thewin, 2020-11-27:
If the send fails, it should return a negative value, like POSIX write(2).  See https://docs.aws.amazon.com/embedded-csdk/202011.00/lib-ref/libraries/standard/coreMQTT/docs/doxygen/output/html/mqtt_transport_interface.html

If a negative value is returned, then `totalBytesSent` is set to the negative value, and the `while` loop exits.
https://github.com/FreeRTOS/coreMQTT/blob/de4edcd578257d34cf061b21f29d0be79720861f/source/core_mqtt.c#L606-L611

The three outcomes are: `totalBytesSent` is negative for an error, is 0 which amounts to a no-op, or is equal to `bytesToSend` for the success case.

snap40ben, 2020-11-30:
Ok, indeed I was missing this. Thanks

dan4thewin, 2020-11-30:
de nada

closed 2020-11-30 by dan4thewin