FreeRTOS/coreMQTT #162: MQTT Subscribing for published data (Payload)

Abhi2109, 2021-04-20:
Hi,
I have a query , after using the AWS library can i get the data which i have sent to AWS IOT core using MQTT?

Lets say i sent {"Temp":80} now as soon as sent the data , i want to receive once it got published using MQTT.

mukesh-2538, 2021-04-20:
Yes, you can subscribe to the same topic and receive it. please refer this sample code
[https://github.com/aws/aws-iot-device-sdk-embedded-C/blob/main/demos/mqtt/mqtt_demo_basic_tls/mqtt_demo_basic_tls.c](url)

Abhi2109, 2021-04-21:
The link is not opening for me

paulbartell, 2021-04-26:
@Abhi2109 Please try this again: https://github.com/aws/aws-iot-device-sdk-embedded-C/blob/main/demos/mqtt/mqtt_demo_basic_tls/mqtt_demo_basic_tls.c

sukhmanm, 2021-05-03:
Hi @Abhi2109,

I will go ahead and close this issue as it appears there are no recent updates. Please feel free to open a new issue if you have any further questions on this topic. Thank you.

closed 2021-05-03 by sukhmanm