FreeRTOS/coreMQTT #183: MQTT_Connect is a blocking call

pietercardoen, 2021-12-14:
Dear

I am using coreMQTT in an event based embedded architecture and I faced problems with the function MQTT_Connect as it's a blocking function, waiting on the connack message. Due to this blocking behaviour, I never get to the point of processing a TCP receive event which is preventing the connack message to be received.

Why is this function implemented to be blocking?

PS. I currently patched the coreMQTT library to avoid this blocking behaviour. I am willing to share this.

muneebahmed10, 2021-12-16:
Hi @pietercardoen,

I understand that you are referring to [`receiveConnack`](https://github.com/FreeRTOS/coreMQTT/blob/d79fbaccd2a3f2ee1a5b1ba922f2e0b1e0ef1f91/source/core_mqtt.c#L1823) in `MQTT_Connect`, and the call stack goes through [`receivePacket`](https://github.com/FreeRTOS/coreMQTT/blob/d79fbaccd2a3f2ee1a5b1ba922f2e0b1e0ef1f91/source/core_mqtt.c#L1535), [`recvExact`](https://github.com/FreeRTOS/coreMQTT/blob/d79fbaccd2a3f2ee1a5b1ba922f2e0b1e0ef1f91/source/core_mqtt.c#L880), and eventually [`recvFunc`](https://github.com/FreeRTOS/coreMQTT/blob/d79fbaccd2a3f2ee1a5b1ba922f2e0b1e0ef1f91/source/core_mqtt.c#L741), the receive function from the transport interface, where your application blocks. How have you implemented the transport interface? Can it be changed so that new incoming data can be read during the receive call? Could the TCP event be processed in an ISR so it will be ready during the Connect's receive call?

The coreMQTT API in core_mqtt.h is a managed MQTT API. The MQTT connection is not complete until the CONNACK has been received, so coreMQTT blocks for it. If there is no data available on the socket, it is expected for the scheduler to schedule another task for that duration so that coreMQTT will not waste any resources. If the managed API is not applicable to your use case, then there is also a low level serialization/deserialization API in [`core_mqtt_serializer.h`](https://github.com/FreeRTOS/coreMQTT/blob/d79fbaccd2a3f2ee1a5b1ba922f2e0b1e0ef1f91/source/include/core_mqtt_serializer.h) and documented [here](https://freertos.org/Documentation/api-ref/coreMQTT/docs/doxygen/output/html/mqtt_design.html#mqtt_serializers). The serialization API can serialize or deserialize all MQTT packets, and your application can take care of sending or receiving them over the network.

abhidixi11, 2021-12-29:
Hello @pietercardoen please let us know if you have any further questions. Also if you could give details about your patch it would be great. Thanks.

pietercardoen, 2022-01-04:
Dear abhidixi11

Apologies for my late reply. I was on holiday. My patch is quite simple: instead of waiting on the connack message, persist state in the context object. Reception of the connack message is handled in the mainloop.

From 4bbcfe7c9a68c04e6d8a50dc1346c343212ee0a1 Mon Sep 17 00:00:00 2001
From: Pieter Cardoen <pieter.cardoen@intulo.be>
Date: Tue, 7 Dec 2021 10:30:34 +0100
Subject: [PATCH] Change to make MQTT_Connect non-blocking

---
 source/core_mqtt.c                    | 147 +++++++++++++-------------
 source/include/core_mqtt.h            |   4 +
 source/include/core_mqtt_serializer.h |   3 +-
 3 files changed, 78 insertions(+), 76 deletions(-)

diff --git a/source/core_mqtt.c b/source/core_mqtt.c
index 8bbccfb..f0ce23d 100644
--- a/source/core_mqtt.c
+++ b/source/core_mqtt.c
@@ -27,6 +27,8 @@
 #include <string.h>
 #include <assert.h>
 
+#include <stdio.h>
+
 #include "core_mqtt.h"
 #include "core_mqtt_state.h"
 
@@ -261,7 +263,6 @@ static MQTTStatus_t sendPublish( MQTTContext_t * pContext,
  * #MQTTSuccess otherwise.
  */
 static MQTTStatus_t receiveConnack( const MQTTContext_t * pContext,
-                                    uint32_t timeoutMs,
                                     bool cleanSession,
                                     MQTTPacketInfo_t * pIncomingPacket,
                                     bool * pSessionPresent );
@@ -1307,8 +1308,54 @@ static MQTTStatus_t receiveSingleIteration( MQTTContext_t * pContext,
                                                   pContext->transportInterface.pNetworkContext,
                                                   &incomingPacket );
 
-    if( status == MQTTNoDataAvailable )
+    if(pContext->connectStatus == MQTTConnecting)
     {
+        // TODO
+        bool sessionPresent = false;
+
+        /* Read CONNACK from transport layer. */
+        if( status == MQTTSuccess )
+        {
+
+            status = receiveConnack( pContext,
+                                     pContext->cleanSession,
+                                     &incomingPacket,
+                                     &sessionPresent );
+        }
+
+        if( status == MQTTSuccess )
+        {
+            /* Resend PUBRELs when reestablishing a session, or clear records for new sessions. */
+            status = handleSessionResumption( pContext, sessionPresent );
+        }
+
+        if( status == MQTTSuccess )
+        {
+            LogInfo( ( "MQTT connection established with the broker." ) );
+            pContext->connectStatus = MQTTConnected;
+            /* Initialize keep-alive fields after a successful connection. */
+            pContext->keepAliveIntervalSec = pContext->keepAliveIntervalSec;
+            pContext->waitingForPingResp = false;
+            pContext->pingReqSendTimeMs = 0U;
[... 200 lines cut ...]
+    uint32_t connectTimeoutMs;
     bool waitingForPingResp;       /**< @brief If the library is currently awaiting a PINGRESP. */
+    bool cleanSession;
 } MQTTContext_t;
 
 /**
diff --git a/source/include/core_mqtt_serializer.h b/source/include/core_mqtt_serializer.h
index bf3d974..1312fc7 100644
--- a/source/include/core_mqtt_serializer.h
+++ b/source/include/core_mqtt_serializer.h
@@ -105,7 +105,8 @@ typedef enum MQTTStatus
     MQTTNoDataAvailable, /**< No data available from the transport interface. */
     MQTTIllegalState,    /**< An illegal state in the state record. */
     MQTTStateCollision,  /**< A collision with an existing state record entry. */
-    MQTTKeepAliveTimeout /**< Timeout while waiting for PINGRESP. */
+    MQTTKeepAliveTimeout, /**< Timeout while waiting for PINGRESP. */
+    MQTTConnectTimeout
 } MQTTStatus_t;
 
 /**
-- 
2.32.0

Kind regards
Pieter Cardoen

abhidixi11, 2022-01-05:
Thanks @pietercardoen .
If I understand correctly, by main loop, you mean handled by application ? It is still going to be the same behavior, i.e., the application is going to wait till CONN_ACK is received. So what is the purpose of making connect asynchronous ? As @cobusve mentioned earlier, you can't do anything else (like sending publish)  until you receive CONN_ACK. 

>I never get to the point of processing a TCP receive event which is preventing the connack message to be received.

I think we trying understand why you are facing the above problem.  You should able to implement your transport receive function in such way you can process TCP event and then process CONN_ACK.

Thanks.

pietercardoen, 2022-01-05:
Dear abhidixi11

The purpose was indeed to obtain identical behaviour. It doesn't make sense to publish messages before a connack is received.

I was using this in an event based system where events of the TCP socket and events of the MQTT library are processed by one thread. As soon as MQTT_connect was started, this event loop got blocked, not allowing any other event to be processed until the MQTT_connect was finished.

TCP socket events needed to receive the connack message were queued but never handled, resulting in a blocking system.

Main reason for this problem is that I was using our existing coreMQTT/sparkplug implementation as backend for a QT graphical application. The problems I experienced was due to the event handling in Qt.

Kind regards
Pieter

abhidixi11, 2022-01-06:
Thanks for clarification @pietercardoen . So basically you are using it in a single threaded system where TCP and MQTT and any other events are processed in a single thread. 

While (1)
{
MQTTSend() // Send is direct does not need event
TCPIPReceive() // check and process incoming TCP /IP /UDP packets the TCP/IP stack receive portion is implemented in this fucntion
MQTTReceive() // check and process incoming MQTT. 
}

So without calling TCPIPReceive() you wont's able to call MQTTReceive().

Our current design assumes TCP/IP packets will be processed in a separate thread.

abhidixi11, 2022-01-06:
Thanks for confirming. We will consider your suggestion for future major release.  I'm closing the issue, please feel free to open a new one if you have any further questions. Thanks!

closed 2022-01-06 by abhidixi11