FreeRTOS/coreMQTT #231: MQTTEventCallback_t  doesn't have user data parameter

ademyankov, 2022-11-17:
A **user** callback is supposed to have **user** data passed in, right?
```
typedef void (* MQTTEventCallback_t )( struct MQTTContext * pContext,
                                       struct MQTTPacketInfo * pPacketInfo,
                                       struct MQTTDeserializedInfo * pDeserializedInfo );
```
There is no way to use user context inside the callback in this form. How would one pass a class object or container of subscriptions? Global variables or static class members are a very lousy approach.

jasonpcarroll, 2022-11-18:
Hi @ademyankov,
The event callback is called by the coreMQTT library when an MQTT packet is received. It isn't used to pass data to coreMQTT- it's used to output information about the packet received and the user defines what they want to do with the packet data from the callback (e.g. check if coreMQTT received a SUBACK packet for a previous call to MQTT_Subscribe).
Best,
Jason Carroll

ademyankov, 2022-11-18:
Ok, don't want to be a jackass, so I will explain.

This is how a "good" user/application callback is supposed to be designed:

```
    typedef void(*MQTTEventCallback_t) (
        struct MQTTContext* pContext,
        struct MQTTPacketInfo* pPacketInfo,
        struct MQTTDeserializedInfo* pDeserializedInfo,
        void* pApplicationData);
```

Then somewhere in a bunch of code, I have my objects constructed, and I call the MQTT init function. Assume that `MQTT_Init` function has the last argument `void* pApplicationData`:

```
    // ...

    auto client = new MyMqttClient();

    // ...

    auto status = MQTT_Init(
        client->context,
        client->transport,
        getTime,
        onEvent,
        client->fixedmem,
        static_cast<void*>(client));
```

(*) Please, take the code above with a huge grain of salt :) (it's just a simple example) because it's an ugly C++ design. `MQTT_Init` is supposed to be called inside of the `MyMqttClient` constructor or inside of some member function like: `MyMqttClient::connect(...)`

Anyway, then in the callback, I can access the MyMqttClient object without any hassle!

```
void onEvent(
    struct MQTTContext* ctx,
    struct MQTTPacketInfo* pi,
    struct MQTTDeserializedInfo* di,
    void* appdata)
{
    // ...

    auto cli = static_cast<MyMqttClient*>(appdata);

    // Example 1: get SUBACK for a subscription
    uint8_t* pCodes{};
    size_t numCodes{};

    MQTT_GetSubAckStatusCodes(pi, &pCodes, &numCodes);
    for (size_t i = 0; i < numCodes; i++) {
        if (pCodes[i] != MQTTSubAckFailure) {
            // Acknowledge that the broker accepted the subscription
            cli->ack(i);
        }
    }

    // ...

    // Example 2: got PUBLISH for a subscription
    if (di->deserializationResult == MQTTSuccess) {
        string topic(di->pPublishInfo->pTopicName, di->pPublishInfo->topicNameLength);
        buffer data(di->pPublishInfo->pPayload, di->pPublishInfo->payloadLength);

        cli->sub[topic].process(data);
    }
}
```

Thus, long story short. Without `pApplicationData` I will have to invent some potentially buggy way of accessing my (user/application) context from within the callback using static members or global pointers/variables or whatever.

I hope I made it clear enough.

Cheers!

Skptak, 2022-11-23:
Hey @ademyankov thanks for bringing this to our attention. I've passed this feature request along to the team in charge of coreMQTT.  In the meantime have you taken a look at the [coreMQTT-Agent Demo](https://www.freertos.org/mqtt-agent/mqtt-agent-demo.html)? This agent has an [init function](https://github.com/FreeRTOS/coreMQTT-Agent/blob/main/source/core_mqtt_agent.c#L974) that takes in a void pointer "pIncomingPacketContext", which seems like the feature you are looking for?
In the meantime if you successfully port this feature to coreMQTT, your contribution back to this repository would be greatly appreciated.

ademyankov, 2022-11-23:
@Skptak hey, no, I haven't seen this agent demo. I will check it out! Thanks a lot.

closed 2022-11-23 by ademyankov