FreeRTOS/coreMQTT #238: TransportInterface_t.writev is no longer optional and must be provided?

jondurrant, 2022-12-23:
Been trying to port my code to v2.1.1 and getting crashes. It looks like the cause is that TransportInterface_t.writev has to be provided to MQTT_init. This used to be optional on v1.2.0.  Documentation does not mention this as now mandatory.

rawalexe, 2022-12-23:
I believe this is because our previous version didn't initialize the `TransportInterface_t` struct at declaration. There is a pull request open to fix it that can be found [here](https://github.com/aws/aws-iot-device-sdk-embedded-C/pull/1845).  Please let us know if your problem is not fixed after the request is merged.

rawalexe, 2022-12-23:
The problem has been patched with the new [PR](https://github.com/aws/aws-iot-device-sdk-embedded-C/pull/1845). I'm closing the issue but if it persists please do reopen it.

closed 2022-12-23 by rawalexe