FreeRTOS/coreMQTT #256: coreMQTT context does not allow user parameters to be passed to the Event Callback

TunaBicim, 2023-07-20:
Hello, 
I have been using the library on a C++ project and would like to pass the this pointer to the Event Callback to be able to access my class instance. 
Is it possible to add a void * pUserData field to the context and pass it to the callback? 
Thank you!

ActoryOu, 2023-07-26:
Hi @TunaBicim,
Thanks for your feedback.
We're discussing this request internally and we'll bring back more info to you.

Thanks!

ActoryOu, 2023-07-27:
For reference, the [post](https://forums.freertos.org/t/coremqtt-how-to-pass-user-data-to-eventcallback/17843) from FreeRTOS forum provides a temp solution for this scenario.
We'll keep this issue opened and plan to support user context in the callback.

Thanks.