FreeRTOS/coreMQTT #259: Wifi Disconnection Handling

genie-ltran, 2023-08-30:
I currently have a task that handles all coreMQTT access for thread safety which I wrote before I learned of coreMQTT agent.  I have a another task that establishes the TLS connection and passes requests for subs, pubs, unsubs, etc.  It's been working well in almost all aspect except for when an unexpected disconnection occurs.

When the wifi disconnection happens, I end the connection establishing task.  Before ending this task I clean everything up (free memory, disconnect calls, etc).  I restart this same task, but the problem a successful connection seems to take a *long* time (very infrequently) or never at all (most of the time).

So is there a guide on how a reconnection should happen on unexpected network disconnection?  I know this question isn't directly related to coreMQTT itself, but I think I've exhausted all the resources available to me.  Thank you.

kar-rahul-aws, 2023-08-30:
Hi @genie-ltran 
Thank you for reporting the issue.
I do not have any guide as of now, but I will take a look if a document exists.
Can you please take a look at these issues [#46](https://github.com/FreeRTOS/iot-reference-esp32c3/issues/46) and [#47](https://github.com/FreeRTOS/iot-reference-esp32c3/issues/47) and check if the [solution ](https://github.com/FreeRTOS/iot-reference-esp32c3/issues/46#issuecomment-1687351129)described here solves your problem.

kar-rahul-aws, 2023-09-05:
Hi @genie-ltran 
Any updates on this issue? Did the fixes suggested solve your problem?

AniruddhaKanhere, 2024-04-10:
Closing this issue due to lack of response.

closed 2024-04-10 by AniruddhaKanhere