FreeRTOS/coreMQTT #266: No new tagged version release for over a year

schoeler, 2023-11-09:
Hi,
I'm wondering why there hasn't been an official release in over a year?  We took v2.1.1 considering it was stable, but when the all the fixes going for keep alive handling going to be merged into an official release?  Some of these things have been fixed for quite some time!
Thank you!

cobusve, 2023-11-10:
I think what happened is just that 2.1.1 was more stable than we anticipated, so there has not been much urgency in doing the work to make another formal release. As it stands we are approaching the end of the previous LTS version support period and in the next couple of months we will be working on getting everything ready for the next LTS version of all our libraries.

This means that, while I can not give you a specific date just yet, we will be making a new tagged version very soon and no later than March 2024 when the current LTS will reach its end.

mavsw, 2023-11-15:
I'm confused, the latest release does not update the timestamp of the last packet received. Which means the code is sending a _MQTT_Ping_ for every _MQTT_ProcessLoop()_

This was fixed here (2022-12): https://github.com/FreeRTOS/coreMQTT/pull/235

Maybe I'm misunderstanding but that means you can not realistically have any long open connections opened with the latest version of coreMQTT, which also is part of the current release of FreeRTOS-LTS. Or is there a workaround I'm not aware of?

I'm seeing a constant stream of ~50 packets/s, no idea what the added cost will be for those of us using cloud resources.

AniruddhaKanhere, 2024-04-10:
Hello @mavsw, I do not have a date for you, but we are working on testing the library before we release it soon!

Stay tuned.

Thanks

moninom1, 2024-05-06:
Released the coreMQTT [v2.2.0](https://github.com/FreeRTOS/coreMQTT/tree/v2.2.0)

closed 2024-05-06 by moninom1