FreeRTOS/coreMQTT #267: Support for MQTT v5.x

srnae05, 2023-11-13:
The AWS MQTT broker supports MQTT v5.x. To make the most out of integration of devices to AWS MQTT broker using MQTT v5.x, the coreMQTT library needs an update as it currently is compliant to MQTT v3.x
Could this update be considered ?

Thank you.

cookpate, 2023-11-21:
MQTT 5 is on our roadmap; however, we don't think anybody is working on it right now.
We would be grateful for any contributions made to accelerate its delivery.

jnz86, 2024-01-06:
Checking in and interested. None of the features look all too difficult. But... Without pointing fingers, a lot of the CoreMQTT code is written in a fairly complex way and I suspect almost no one is going to tackle this on their own. 

If someone at Amazon were to scaffold out the required hooks, it seems much more likely for independent contributors to jump in.

amazonKamath, 2024-07-21:
@srnae05  @jnz86  we started working on MQTTv5.  Here is the draft PR https://github.com/FreeRTOS/coreMQTT/pull/301.  It is not ready yet, and only supports connect, publish and disconnect. Subscribe is not implemented.  It would be great if you can help test it out for your use cases and give us some feedback? Thank you.

DakshitBabbar, 2025-02-10:
This issue is very similar to #269 which contains more details and ongoing discussion. To keep things consolidated, I’m closing this issue. Please follow that thread for updates and feel free to add any additional insights there.

Thanks for your contribution!

closed 2025-02-10 by DakshitBabbar