FreeRTOS/coreMQTT #270: No ack found for packet id

BogdanTheGeek, 2023-12-15:
After a running for a long time, these messages accumulate:
```
I (146451442) network: Attempting to connect to MQTT
I (146450881) hmqtt: Connected to MQTT.
I (146450882) hmqtt: Session present.
E (146450882) coreMQTT: No ack found for packet id 15788.
E (146450886) coreMQTT: No ack found for packet id 16425.
E (146450893) coreMQTT: No ack found for packet id 60272.
E (146450899) coreMQTT: No ack found for packet id 26707.
E (146450905) coreMQTT: No ack found for packet id 28396.
E (146450912) coreMQTT: No ack found for packet id 49384.
E (146450918) coreMQTT: No ack found for packet id 49987.
E (146450925) coreMQTT: No ack found for packet id 62338.
E (146450931) coreMQTT: No ack found for packet id 43880.
E (146450938) coreMQTT: No ack found for packet id 34763.
I (146450944) hmqtt: Session resumed.
I (146451992) network: ConnectToMQTT succeeded
I (146450954) hmqtt: Waiting for network to be ready...
I (146450960) network: Subscribing to MQTT topics ...
```
This only happens on some devices, some of the time.
What is the cause of this issue? Is it due to maintaining the session across reconnects?

I am using coreMQTT v2.1.1 and coreMQTT-Agent V1.2

bradleysmith23, 2023-12-15:
Hello, @BogdanTheGeek! This would be a great forum question, thank you for the question! Could you repost this question on the [FreeRTOS Forums](https://forums.freertos.org/)? It would get a lot more discussion over there (You can sign in with your Github account). Thank you!

closed 2023-12-15 by bradleysmith23

BogdanTheGeek, 2023-12-16:
Thank you @bradleysmith23.
Here is the conversation for anyone interested in the future:
https://forums.freertos.org/t/coremqtt-no-ack-found-for-packet-id/18737