FreeRTOS/coreMQTT #277: What happens when the monotonic clock overflows?

davidchisnall, 2024-03-04:
The monotonic clock uses a 32-bit integer in milliseconds and so will overflow in just under 50 days.  It does not have a per-connection epoch, so the integrator is required to use something like from boot or from first connection.  After 50 days, it will overflow.  Does the library handle this (in which case, can it be documented)?  Is there some way of resetting the epoch?  Failing that, is it possible to have an option to use 64-bit timestamps?

archigup, 2024-03-05:
Hey, thanks for reaching out! I agree this could be more clear.

It is currently documented here: https://github.com/FreeRTOS/coreMQTT/blob/9a3edb6b67195c9aefb62be4372c33e97f2342c4/source/include/core_mqtt.h#L257

The time function is only used for calculating durations for timeouts and keep alive periods.

Overflow is not an issue here, as unsigned wraparound is defined, and will return correct results when subtracting, as long as they are less than half the range apart. Since the durations we are measuring are on the order of seconds, and not days, this is fine.

davidchisnall, 2024-03-05:
Thanks. As long as the operations are always add and subtract on that value and you never do ordered comparisons on the result then that should be fine. That’s the kind of constraint that it’s easy for someone to forget, so it would be good to document that explicitly in the callback type.

AniruddhaKanhere, 2024-04-11:
@davidchisnall we have added a bit more documentation to the codebase in the above PR https://github.com/FreeRTOS/coreMQTT/pull/279.

Closing this thread now.

Let us know if you find our solution insufficient or lacking and feel free to reopen the thread.

Thanks,
Aniruddha

closed 2024-04-11 by AniruddhaKanhere