FreeRTOS/coreMQTT #278: Is it safe to call coreMQTT APIs from a user callback?

hlef, 2024-04-10:
Hi,

Is it safe for a [user callback](https://freertos.github.io/coreMQTT/v2.1.1/group__mqtt__callback__types.html#ga00d348277ed4fde23c95bfc749ae954a) to call a coreMQTT API?

I seem to be seeing that calling `MQTT_ProcessLoop` from a user callback wouldn't be safe if the nested call hits an error path and calls `discardStoredPacket` ([here](https://github.com/FreeRTOS/coreMQTT/blob/main/source/core_mqtt.c#L1755)), setting the `index` to 0 ([here](https://github.com/FreeRTOS/coreMQTT/blob/main/source/core_mqtt.c#L1171)), and causing an overflow ([here](https://github.com/FreeRTOS/coreMQTT/blob/main/source/core_mqtt.c#L1785)), but I hope to be wrong.

AniruddhaKanhere, 2024-04-10:
Hello @hlef,

The user callback is called from within `MQTT_ProcessLoop`. So, if you call coreMQTT APIs from within the callback, it might cause a nested calling of `MQTT_ProcessLoop` as `MQTTProcessLoop -> callback -> MQTT_ProcessLoop ...` which might lead to increased stack consumption in the least.

And in the worst case, as you pointed out, there can be data corruption as one call to ProcessLoop does one thing, whereas the nested call changes a value in use by the parent call.

The coreMQTT library is not designed to be thread safe to keep it simple and light weight. It is recommended to allow only one thread to call MQTT APIs or to protect them with mutex if more than one threads are going to be calling the APIs simultaneously.

hlef, 2024-04-11:
Hi Aniruddha,

Thanks for the quick answer. We have a scenario where we want to do a publish in response to an incoming publish event.

It is quite convenient to do the publish in the incoming publish callback. If that is not safe we will set some flags and perform publishes once we are out of the process loop.

Thanks!

AniruddhaKanhere, 2024-04-11:
Hello Hugo,

Thank you for that information!

Yes, it will not be safe to call MQTT API from the callback itself. As you mentioned, setting some flags would be helpful. Ideally, I would not put too many things in the callback. I'd much rather have the handler copy data into a buffer or put it in a queue (both options have their pros and cons) which can be used to 'kick' a different DataProcessing task into action to process the data.

I say that because doing multiple things in the callback might cause MQTT ProcessLoop not being called enough. Additionally, the processing of the data will happen at the priority of the task calling ProcessLoop - you may or may not want that. Having a different task will let you process the data when other high priority tasks are sleeping or blocked.

But, of course I do not know your whole design - so take my comments as 'best practices' instead of something you must do.

That being said, seems that I have answered your question and will be closing this issue. If that is not the case, feel free to re-open the thread (or create a new one). :)

Thanks,
Aniruddha

closed 2024-04-11 by AniruddhaKanhere

hlef, 2024-04-18:
That answers my question, thanks! I think that it would be helpful to mention some of this in the documentation of the [user callback type](https://freertos.github.io/coreMQTT/v2.1.1/group__mqtt__callback__types.html#ga00d348277ed4fde23c95bfc749ae954a) (for example as a note).