FreeRTOS/coreMQTT #280: Fix topic name parsing

facucc, 2024-04-12:
The value of the topic name variable is equal to topic name + payload
The error is in the [line 1448](https://github.com/FreeRTOS/coreMQTT/blob/main/source/core_mqtt_serializer.c#L1448)

AniruddhaKanhere, 2024-04-14:
Hello @facucc,

Thank you for reaching out to us.

The topic name pointer (`pPublishInfo->pTopicName`) returned by the code is NOT NULL terminated. Your application must use the topic name length (`pPublishInfo->topicNameLength`) to determine where does the topic end.

This is done to reuse the existing payload memory instead having to copy it to a different string in order to NULL terminate it.

Does that answer your question? Or are you saying that the topic name length variable is set incorrectly so that you are reading into the payload?

Thanks,
Aniruddha

facucc, 2024-04-14:
Hi @AniruddhaKanhere, If I want to use strcmp it does not work and may generate errors, I fixed it locally, if you want I will share the change that should be made, because if I want to use the topic name variable it doesn't work very well

AniruddhaKanhere, 2024-04-15:
Hello @facucc,

I take it by your 'thumbs up' reaction that this fixed the issue you were having. Thus, I'll be closing this ticket.

Please feel free to reopen this thread if I misunderstood you and the issue persists.

Thanks,
Aniruddha

closed 2024-04-15 by AniruddhaKanhere