FreeRTOS/coreMQTT #282: Compile error for comparison of integer expressions of different signedness

skyfishyubo, 2024-04-24:
../../../../lib/mqtt_client/coreMQTT/source/core_mqtt.c: In function 'sendConnectWithoutCopy':

../../../../lib/mqtt_client/coreMQTT/source/core_mqtt.c:2220:50: error: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Werror=sign-compare]

         assert( ( pIndex - connectPacketHeader ) <= sizeof( connectPacketHeader ) );

moninom1, 2024-04-25:
Hello @skyfishyubo  ,
Thank you for bringing this to our attention! We will take a look at this issue. Thank you for your patience!

tony-josi-aws, 2024-04-25:
@skyfishyubo 
Closing this issue as #283 should fix this. Please feel free to reopen if you think otherwise.

closed 2024-04-25 by tony-josi-aws