FreeRTOS/coreMQTT #287: PING message interval is not handled properly

ademyankov, 2024-06-01:
It looks like there is a bug in the line `core_mqtt.c:1383`

            if( ( timeElapsed != 0U ) && ( timeElapsed >= PACKET_RX_TIMEOUT_MS ) )

By default `PACKET_RX_TIMEOUT_MS` is 30 seconds but `PING` message is sent every time the Loop function is invoked because `timeElapsed` is always equal to `now` when `pContext->lastPacketRxTime` is `0`

The fix I did is this one but I am not sure how it might effect other parts of the code:

            static uint32_t calculateElapsedTime( uint32_t later, uint32_t start )
            {
                return start ? later - start : 0;
            }

I haven't checked other conditions related to invoking `MQTT_Ping( pContext );`

paulbartell, 2024-06-03:
@ademyankov Thanks for pointing this out. I'll get it over to our coreMQTT experts.

Pooja-Toshniwal, 2024-06-05:
Which version of coreMQTT are you using? @ademyankov

ademyankov, 2024-06-05:
@Pooja-Toshniwal v2.1.1

Pooja-Toshniwal, 2024-06-05:
After investigating, it appears that the problem you encountered with unnecessary MQTT_Ping invocations is present in the older version (2.1.1) of the FreeRTOS coreMQTT library. This issue has been resolved in the newer version (2.2.0) by [introducing](https://github.com/FreeRTOS/coreMQTT/blob/238350a844336c6dea7e3aa50ecab729bf27e288/source/core_mqtt.c#L1794) an additional condition check that updates the `lastPacketRxTime` whenever an incoming packet is handled (except for CONNACK). By updating `lastPacketRxTime` when an incoming packet is received, the library ensures that the elapsed time calculation is correct, even in cases where no packet has been received initially. This change resolves the issue of unnecessary `MQTT_Ping` invocations when`lastPacketRxTime` is zero.
This condition has been added- 
`if( status == MQTTSuccess )
        {
            pContext->lastPacketRxTime = pContext->getTime();
        }`
https://github.com/FreeRTOS/coreMQTT/blob/238350a844336c6dea7e3aa50ecab729bf27e288/source/core_mqtt.c#L1794

I suggest updating to the latest version of the library ( v2.2.0)  and testing it to confirm its functionality. If you still encounter any issues, please let us know. @ademyankov

ademyankov, 2024-06-05:
@Pooja-Toshniwal yes, I've just checked v2.2.0 and it works as excpected. Thank you for investigating the problem.

closed 2024-06-05 by ademyankov