FreeRTOS/coreMQTT #292: Hardfault issue

pistoletov1974, 2024-06-19:
https://github.com/FreeRTOS/coreMQTT/blob/a2459c6317077c166782d9d79ce5c2c97058bce3/source/core_mqtt.c#L796
when writev function doed not implement due to the stack being painted to the value a5, the function is called at the address a5a5a5a5, which leads to HardFault

aggarg, 2024-06-19:
Am I right in understanding that you `writev` member of the transport interface is not initialized and that leads to hard fault. If yes, that is an application bug. Why are you not initializing the `writev` member?

pistoletov1974, 2024-06-19:
> Why are you not initializing the `writev` member
 * @note If the proper write functionality is not present for a given device/IP-stack,
 * then there is no strict requirement to implement write.
 I initialize with 
`        xTransport.send = Plaintext_FreeRTOS_send;
        xTransport.recv = Plaintext_FreeRTOS_recv;
        xTransport.writev = 0x0UL;     `
and issue was resolved. Before adding xTransport.writev = 0x0UL the app halted.

aggarg, 2024-06-19:
Thank you for explaining.

> Before adding xTransport.writev = 0x0UL the app halted.

The only invalid value that we can reliably check a pointer against is NULL which is checked [here](https://github.com/FreeRTOS/coreMQTT/blob/a2459c6317077c166782d9d79ce5c2c97058bce3/source/core_mqtt.c#L794).  Passing uninitialized/garbage value for `writev` member seems to be application bug and I do not think we can (or even we should attempt to) differentiate between a valid pointer value or a garbage value. Your fix seems correct and there is nothing else to do here - or am I missing something?

aggarg, 2024-06-19:
I closing this issue. Let us know if you need anything else.

pistoletov1974, 2024-06-19:
> pointer against is NULL which is checked [here](https://github.com/FreeRTOS/coreMQTT/blob/a2459c6317077c166782d9d79ce5c2c97058bce3/source/core_mqtt.c#L794)

Yes, you are right, I assigned 0 just to pass this check. I believe that if you do not initialize this value it will not be NULL, it will most likely be 0xA5A5A5. This is a special constant that the kernel fills the heap to determine the maximum usable heap size. The so-called memory painting method. The solution to this problem is to assign 0x00UL as default value  if you do not implement this function

kstribrnAmzn, 2024-06-19:
> I believe that if you do not initialize this value it will not be NULL

This is true in your case because the FreeRTOS kernel initializes the stack to the 0xA5A5... value and the transport interface structure resides in the stack. If the transport interface structure was entirely initialized to 0x0UL (or resided in memory that was initialized to 0x0UL) then you would not have faced this error.

As @aggarg mentioned the code cannot deterministically rule out that 0xA5A5... is invalid, only that `NULL` is. Your fix was to set the `writev` pointer explicitly, which is the right thing to do. There isn't further work to be done here.