FreeRTOS/coreMQTT #296: Invalid git tag for V2.3.0

BogdanTheGeek, 2024-07-09:
[image]
Cannot checkout v2.2.0 or v2.3.0 as the tags don't belong to this repo.

BogdanTheGeek, 2024-07-09:
Also, the runtime version has not been update:
https://github.com/FreeRTOS/coreMQTT/blob/8b3abb70365cdedcb5a12f0aa8639458b89aa3b1/source/include/core_mqtt.h#L48

chinglee-iot, 2024-07-09:
@BogdanTheGeek 
Thank you for reporting the MQTT_LIBRARY_VERSION problem. We will fix this issue. As for the tag problem, the release is created from a branch. The branch will be deleted after the tag release, but the tag and commit will still exist.

BogdanTheGeek, 2024-07-10:
@chinglee-iot It might be so, but that still means I can't checkout the tag. The tag should be created after merging into main.

chinglee-iot, 2024-07-15:
@BogdanTheGeek 
After the branch is deleted, the commits referenced by the tag remain valid. Can you share us the error message you encountered when checking out the tag?

moninom1, 2024-07-22:
HI @BogdanTheGeek, Can you please help us know what error you are getting when checking out the tag?

ActoryOu, 2024-07-29:
Hi @BogdanTheGeek,
I can clone tag v2.3.0 successfully. Could you help provide the error message you met for futher analysis?

My clone log:

> PS C:\git\temp> git clone -b v2.3.0 https://github.com/FreeRTOS/coreMQTT
Cloning into 'coreMQTT'...
remote: Enumerating objects: 7539, done.
remote: Counting objects: 100% (2288/2288), done.
remote: Compressing objects: 100% (367/367), done.
Receiving objects: 100% (7539/7539), 4.70 MiB | 2.26 MiB/s, done.eused 5251
Resolving deltas: 100% (5737/5737), done.
Note: switching to '3f4cedc1ad027d76b2f00c2f83481b212b7a8632'.

Thank you.

ActoryOu, 2024-08-02:
This issue is being closed as the reported behavior could not be reproduced in other environments after thorough testing. Nevertheless, if you continue to experience the same issue, feel free to reopen this ticket.