FreeRTOS/coreMQTT #310: receive  and send packet separate thread

skyfishyubo, 2024-10-31:
receiveSingleIteration have handleKeepAlive function. could we relealize a non-static handleKeepAlive function so that could be call in TX thread, in this way we could sleep for wait to receive message, and have different thread to handle the keepalive issue.

ActoryOu, 2024-10-31:
Hi @skyfishyubo,
Thank you for your suggestion about separating receiveSingleIteration and handleKeepAlive. Could you please elaborate on the benefits you see in this approach? We're interested in understanding your perspective to better evaluate this potential change. Your insights would be greatly appreciated.

Thank you.

archigup, 2024-11-01:
Hey, if you use MQTT_ProcessLoop, it handles keep-alives. You can instead use MQTT_ReceiveLoop, which doesn't handle keep-alives, allowing you to handle them yourself. You could then send them on another thread on a schedule, or however else you would like.

closed 2024-12-03 by archigup