FreeRTOS/coreMQTT #317: Express the need of a non-blocking TransportRecv_t

diopoex, 2025-04-03:
I think the documentation could be made more clearly on these two cases:

[image]
https://freertos.github.io/coreMQTT/v2.3.1/group__mqtt__callback__types.html#ga227df31d6daf07e5d833537c12130167

I think there could directly be mentioned that if the transport Interface is blocking the keepAlive mechanism isnt working, because usually blocking sockets only return 0 on recv() when the peer has closed the connection(see: https://github.com/FreeRTOS/coreMQTT/blob/8ec72d6ef7d61c78d899f069395ac5fed7624650/source/core_mqtt.c#L1769).

Then a mention in the "Porting Guide" and/or in the "Transport Interface" could also make sense.

diopoex, 2025-04-03:
If this makes sense I will gladly make the changes and create a pull request.

rohitmadan07, 2025-04-07:
Thanks for your contribution ! The PR has been merged. Closing this issue

closed 2025-04-07 by rohitmadan07