FreeRTOS/coreMQTT #324: When the server returns a SUBACK, the QoS level requested by the client in the SUBSCRIBE message can be changed, but the MQTT_DeserializeAck interface cannot retrieve the QoS level returned in the SUBACK.

Ryan-CW-Code, 2025-06-05:
Client requests QoS: When the client sends a SUBSCRIBE message, it specifies a requested QoS level (0, 1, or 2) for each topic it wishes to subscribe to8.
Server Authorization QoS: When a server receives a SUBSCRIBE message and processes the subscription request, it determines the highest QoS level to grant for that subscription based on its own policies, configurations, permission settings, or capability limitations. This granted level can be the same as the one requested by the client or potentially lower.
Return authorized QoS via SUBACK: In the payload section of the SUBACK message sent to the client, the server includes a return code list. Each byte in this list, in order, corresponds to a topic subscription entry in the SUBSCRIBE message, with its value explicitly indicating the actual QoS level granted by the server for that topic (0, 1, 2) or a failure code representing a subscription failure (typically 0x80)811.
For example, the client requests to subscribe to the topic sensor/temp and specifies QoS 2.
The server may return 1 in the payload of SUBACK if it only supports the highest QoS 1 due to its own configuration.
Or, if the server supports QoS 2 and authorization is successful, it returns 2.

DakshitBabbar, 2025-06-06:
Hey @Ryan-CW-Code,
The payload of the packet received can be accessed through the `MQTT_DeserializeAck` API. You can try to read the status codes of each of the topics like done in this [serializer demo ](https://github.com/FreeRTOS/FreeRTOS/blob/c4da4ef1a7c469b55070d63a0cfd68255ff09e65/FreeRTOS-Plus/Demo/coreMQTT_Windows_Simulator/MQTT_Serializer/DemoTasks/SerializerMQTTExample.c#L1270C17-L1270C42). You can also use the [`MQTT_GetSubackStatusCodes`](https://github.com/FreeRTOS/coreMQTT/blob/42d843f40ef2abc752a419f45f7adb7dac67e591/source/core_mqtt.c#L3595) API as used in [this plaintext demo](https://github.com/FreeRTOS/FreeRTOS/blob/c4da4ef1a7c469b55070d63a0cfd68255ff09e65/FreeRTOS-Plus/Demo/coreMQTT_Windows_Simulator/MQTT_Plain_Text/DemoTasks/PlaintextMQTTExample.c#L707).

Try the above methods and let us know if you face any issues.

Ryan-CW-Code, 2025-06-06:
Thank you for your reply, indeed it is possible to achieve the function I want by judging the payload data at the application layer
But this method has poor maintainability.
Perhaps I was wrong, when the server detects that the subscribed QoS topic is not supported, the `SUBACK` will only return 0x80?
Please note that I only used `core_mqtt_serializer.c` and `core_mqtt_serializer.h`

DakshitBabbar, 2025-06-09:
You're absolutely right — it's entirely expected behavior for the server to grant a QoS level that is lower than the one requested. Regarding the processing of the SUBACK payload, it would not be appropriate for the library to handle this (especially on a per-topic basis), as it does not internally track which topics have been subscribed to. This responsibility rightly belongs to the application layer, which maintains knowledge of the original subscription requests.

Another key reason for not incorporating this feature into the library is its intended use on resource-constrained devices. Parsing the SUBACK payload and managing granted QoS levels for each topic is not a universal requirement and could introduce unnecessary overhead for users who prefer minimal processing. 

Let us know if you have any questions.

Ryan-CW-Code, 2025-06-09:
Have you considered handling suback the same way as publish.
Handle it separately using the `MQTT_DeserializePublish` function.
We can use a similar approach, such as extracting the suback parsing from `MQTT_DeserializeAck` and using a separate function `MQTT_DeserializeSuback`, which should not significantly increase ROM usage

Ryan-CW-Code, 2025-06-11:
Sorry, I didn't describe my question clearly.
Because the server may modify the `QoS` level of the subscription topic when replying with `SUBACK`, I want to obtain the real `QoS` level authorized by the server when parsing the `SUBACK` packet.
Currently, `coreMqtt` uses the `MQTT_DeserializeAck` API to process `SUBACK` messages. The return value of the `MQTT_DeserializeAck` API does not include the server's authorized real `QoS` level; the `MQTT_DeserializeAck` API only returns the parsing result and the original message header information.
If I need to re-parse based on the original message header information to obtain the true `QoS` level authorized by the server, this is not good.
Because the `MQTT_DeserializeAck` API has already parsed the server's authorized real `QoS`, it is just that the real `QoS` authorized by the server cannot be obtained through return values and other information.

Ryan-CW-Code, 2025-06-13:
Thank you for Owner's answer!
I will temporarily close the issue; I have temporarily encapsulated an `MQTT_DeserializeSuback` function to implement the functionality.
Let's discuss this issue once the `coreMqtt v5` version is supported.
Thank you again.

closed 2025-06-13 by Ryan-CW-Code