FreeRTOS/coreMQTT #328: incomingPacketValid add more validation

Ryan-CW-Code, 2025-09-15:
incomingPacketValid needs more validation to avoid misidentifying data that does not meet the specifications
[image]

Ryan-CW-Code, 2025-09-15:
Although increased judgment cannot completely avoid misidentifying data, it will help.

AniruddhaKanhere, 2025-09-18:
Hello,

I see what you meant. But validating the remaining bits will not protect against these kinds of scenarios. It'll make it a bit less likely to happen.

That being said, MQTT requires that the underlying network channel is lossless and ordered stream. Such as TCP. TCP will take care that any missing data is received from the server before handing the packet to the application. Lossy network protocols such as UDP are not viable for MQTT.

For reference, see this section about the network from the official specification: https://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html#_Toc398718098.

Ryan-CW-Code, 2025-09-22:
You are correct! I ignored the fact that TCP ensures the integrity and order of data packets through a reliable transmission mechanism.
Thank you again.

closed 2025-09-22 by Ryan-CW-Code